AllenNeuralDynamics/aind-fip-dff #2: Review DF/F of existing capsule for scientific validity

jeromelecoq, 2024-07-02:


jeromelecoq, 2024-07-02:
We should bring Johannes into this meeting.

hagikent, 2024-07-03:
To my understanding, the last iteration (March/April) was as follows: 
-Stefano + I identified that Johannes's preprocessing is very harshly subtracting slow baseline signal 
-the reason seemed to be that the simulation process included "blank" recording channel in its learning data.
-Stefano suggested Johannes using only "real" data for setting up simulation.
-not sure if any amendment has been done since then.

j-friedrich, 2024-08-13:
## [Capsule](https://codeocean.allenneuraldynamics.org/capsule/1390547) with the updated simulations.

- Based on Autofluorescence and DeadSensor data the **baseline** is well described by the product of two factors, (1) an Biphasic exponential decay modeling bleaching, and (2) an increasing saturating exponential modeling brightening. 
- **Motion attenuation** is well modeled as colored noise. 
- Based on the 4NM data the **calcium** due to **neural activity** can be modeled as colored noise convolved with an exponentially decaying kernel.

### Together, we have following generative model for FIP data.

[image]

## [Pipeline](https://codeocean.allenneuraldynamics.org/capsule/4871997) converting the outputs into NWB files.
The resulting NWB files are saved as this [data asset](https://codeocean.allenneuraldynamics.org/data-assets/4f6e8057-ffa1-41fe-9ab0-57c71567e8a7)

## [Capsule](https://codeocean.allenneuraldynamics.org/capsule/3443145) comparing the three ΔF/F methods.
- 4th order polynomial fit using ordinary least squares (OLS)
- biphasic exponential fit using OLS: $a\cdot e^{-b\cdot t} + c\cdot e^{-d\cdot t}$
- Robust fit with  Biphasic exponential decay (bleaching)  x  increasing saturating exponential (brightening) using iteratively reweighted least squares (IRLS):  $b_{\infty} \cdot (1 + b_{slow} e^{-t/\tau_{slow}} + b_{fast} e^{-t/\tau_{fast}}) \cdot (1-b_{bright} e^{-t/\tau_{bright}}))$

hagikent, 2024-08-13:
The problem of the previous version was that baseline subtraction was too greedy, resulting in making most of the signals be with complete flat baseline. 
@rachelstephlee and I should test this in the context of tonic/phasic DA signal project.

alexpiet, 2024-08-13:
Testing this new method and its implementation requires resolving (https://github.com/AllenNeuralDynamics/aind-fip-dff/issues/5) first

jeromelecoq, 2024-08-13:
Moving to testing. Separate ticket

closed 2024-08-13 by jeromelecoq