AllenNeuralDynamics/aind-fip-dff #4: nwb.acquisitions reward delivery time does not contain the reward volume

alexpiet, 2024-08-02:


hanhou, 2024-08-08:
> scratch['metadata'] should have all the information in the aind-metadata session.json, right?
Actually the opposite is true. Scratch is my temporary way of storing meta data, so we should remove it and make sure that all information in the scratach exists in the json metadata files.

Ahad-Allen, 2024-08-09:
> * nwb.stimulus should contain:
>       
>       * go Cue (currently only in trials)
>       * rewards (currently in acquisitions, and only has time of rewards, not size)
>         
>         * `data` should contain the reward volume, instead of "1" which is the current value
> 
>     * scratch['metadata'] should have all the information in the aind-metadata session.json, right?
> 
>     * nwb.experimenter is none
> 
>     * nwb.session_description is the string "Test File"
> 
>     * nwb.lab is none
> 
>     * nwb.notes is none (we have notes in the metadata)
> 
>     * nwb.protocol is none
> 
>     * nwb.surgery is none
> 
>     * nwb.virus is none

Most of this makes sense to update! We can pull a lot of the missing metadata from the json files. The scratch section should be removed, as Han pointed out. I'm a little confused on the changes you'd like to do to the stimulus. Are you suggesting we add the Stim table with reward times and go Cues in it(I am not sure what a go Cue is, but that is probably just my ignorance on experimental design here)? That shouldn't be too large of a change, if so!

alexpiet, 2024-08-09:
Yes, we should add a stimulus table with reward times (water delivery, currently in acquisition), and go cue (sound played to the mouse, currently only in the trials table)

Ahad-Allen, 2024-09-03:
@rachelstephlee here's the metadata objects we should be adding

rachelstephlee, 2024-09-03:
where was the session_description set? I can't re-set it. NWB gives me an error that the attribute is already set.

rachelstephlee, 2024-09-10:
@rachelstephlee will followup with @Ahad-Allen on this.

Ahad-Allen, 2024-09-10:
I believe @rachelstephlee already did some work on adding most of this metadata, and I have some code that should move where the gocues are present. We still need to add reward volume and remove scratch. I can work on these now

Ahad-Allen, 2024-09-11:
Good news, the reward volume size is already in the trials table as reward_size_left and reward_size_right. I added the gocues in the feat-4 branch. I'll push a pr now

Ahad-Allen, 2024-09-11:
PR: https://github.com/AllenNeuralDynamics/NWB-Packaging-FiberPhotometry-Base-Capsule/pull/21

alexpiet, 2024-09-11:
Updated specs:
nwb.acquisitions['left_reward_delivery_time'].data should contain reward volume

These fields should be grabbed from the aind metadata files
nwb.experimenter is none
nwb.session_description is the string "Test File"
nwb.lab is none
nwb.notes is none (we have notes in the metadata)
nwb.protocol is none
nwb.surgery is none
nwb.virus is none

alexpiet, 2024-09-11:
https://github.com/AllenNeuralDynamics/NWB-Packaging-FiberPhotometry-Base-Capsule/pull/16
adds `nwb.surgery`, `nwb.Experimenter`, `nwb.protocol`, and `nwb.lab`

alexpiet, 2024-09-12:
Need to define an NWB specification [here](https://alleninstitute-my.sharepoint.com/:w:/g/personal/alex_piet_alleninstitute_org/Eeep60T6VnZBuzW7SA0ujrUBM6mRdNSPVVohIFDJly84Yw?e=MH50ti)

jeromelecoq, 2024-09-12:
Assigning to Alex to finish document. @Ahad-Allen  can handle once done.

alexpiet, 2024-09-17:
This NWB extension is for adding AIND metadata to NWB files
https://github.com/AllenNeuralDynamics/ndx-aind-metadata

alexpiet, 2024-10-16:
Feedback needed:
- [ ] `nwb.session_description` is currently set to "NWB file generated by AIND pipeline". Do we want to add anything more? seems like AIND metadata is sufficient
- [ ] `nwb.identifier` is a long alphanumeric string, it doesn't appear to be a CO asset ID, so I don't know what it is
- [ ] `nwb.timestamps_reference_time`. This is currently set to the session_start_time. But I think this is wrong. The harp timestamps are not reset when the session starts (https://github.com/AllenNeuralDynamics/dynamic-foraging-task/issues/319). I dont know how to set the reference time.

rachelstephlee, 2024-10-16:
would we want a version # of the pipeline added in "NWB file generated by AIND pipeline"? othewise we are guessing when the data is being processed + the state of the pipeline at that time

alexpiet, 2024-10-16:
> would we want a version # of the pipeline added in "NWB file generated by AIND pipeline"? othewise we are guessing when the data is being processed + the state of the pipeline at that time

I think that should be taken care of in the processing metadata

alexpiet, 2024-10-23:
I believe I have finished reviewing how to organize the NWB files, and generate many specific issues. 

Details here: https://alleninstitute-my.sharepoint.com/:w:/g/personal/alex_piet_alleninstitute_org/Eeep60T6VnZBuzW7SA0ujrUBM6mRdNSPVVohIFDJly84Yw?e=dAPmUI

I would like feedback on these specific issues, as well as reviewing the file above:
- [ ] `nwb.session_description` is currently set to "NWB file generated by AIND pipeline". Do we want to add anything more? seems like AIND metadata is sufficient
- [ ] `nwb.identifier` is a long alphanumeric string, it doesn't appear to be a CO asset ID, so I don't know what it is. 
    - Edit: turns out this is a uuid4 created by the NWB_Packaging_Subject capsule (https://github.com/AllenNeuralDynamics/NWB_Packaging_Subject_Capsule/blob/0817b7aa432c788d00c49aab0fa5da19a5199d07/code/run_capsule.py#L284C23-L284C36)
- [ ] `nwb.timestamps_reference_time`. This is currently set to the session_start_time. But I think this is wrong. The harp timestamps are not reset when the session starts (https://github.com/AllenNeuralDynamics/dynamic-foraging-task/issues/319). I dont know how to set the reference time.

Ahad-Allen, 2024-10-25:
Session_Description could maybe be generated using the processing.json to add a little more time?
Not sure on the reference time, perhaps we could ask someone who's worked on Harp on this.

jeromelecoq, 2024-10-29:
session_description should include the task + modalities recorded. 
nwb.timestamps_reference_time is absolute to world clock. 

We agreed to Reset Harp timestamps  at beginning to task for new experiments 
and calculate the value the field should be for past experiments, accepting some imprecision (seconds off is ok).

jeromelecoq, 2024-10-29:
Related issue : https://github.com/AllenNeuralDynamics/dynamic-foraging-task/issues/319

jeromelecoq, 2024-10-29:
@alexpiet will coordinate there.

dyf, 2024-11-05:
has been broken down into sub-issues. more changes incoming once HED tags discussion resolves.

closed 2024-11-05 by dyf