AllenNeuralDynamics/aind-fip-dff #5: FIP timeseries is incorrectly aligned

alexpiet, 2024-08-07:


hagikent, 2024-08-12:
this might be related to the handling of the "10 fist blank frames" that we used to have in the past

alexpiet, 2024-08-12:
Note that debugging this code is very annoying because of this issues outlined here: https://github.com/AllenNeuralDynamics/aind-physio-arch/issues/45#issuecomment-2276796729

rachelstephlee, 2024-08-13:
Todo for rachel: 

Write a testing function that will eventually be integrated into the QC code.

alexpiet, 2024-08-15:
@srecanatesi gives this fix, editing line 460 of "aligned_fiber_to_nwb.py" which reads:
```
        timestamps[channel] = df_fip_sel[df_fip_sel['channel'] == channel]['time_fip'].values
```
to 
```
        timestamps_channel = df_fip_sel[df_fip_sel['channel'] == channel]['time_fip'].values
        conversion_factor = np.round(statistics.mode(np.round(np.diff(timestamps_channel),2))/0.05)
        power = 10**round(np.log10(abs(conversion_factor)))
        timestamps[channel] = timestamps_channel / power

```

This appears to resolve the issue. Remaining tasks:
- [ ] add this line to the pipeline
- [ ] Add a QC check to look for this issue in the future
- [ ] Test on more sessions

alexpiet, 2024-08-15:
- PR to update the pipeline. https://github.com/AllenNeuralDynamics/NWB-Packaging-FiberPhotometry-Base-Capsule/pull/14
- I'm struggling to test it on CO, because I'm not familiar with CO tools.

alexpiet, 2024-08-15:
[image]

hagikent, 2024-08-15:
If easily debugable, I wouldn't discourage to do so. However, practically, we can completely drop the "falling" from nwb or dataframe. 

Rationale:
-rising is corresponding to the starting time of Green channel camera frame exposure
-falling is corresponding to the ending time of Green channel camera frame exposure
-Green channel one data point has only one (Bonsai-software) timestamp and thus only one HARP-aligned timestamp.
-thus, the only one additional info we have by keeping falling is the exposure time (diff between rising and falling)

alexpiet, 2024-08-26:
Making a list of sessions where the alignment is to ISO, and not G1

BAD:
behavior_717531_2024-08-07_09-01-02_processed_2024-08-08_13-34-22 (but still has 30 second offset issue)

GOOD:
behavior_711042_2024-08-23_13-27-08_processed_2024-08-24_07-39-12
behavior_717531_2024-08-23_08-58-49_processed_2024-08-24_10-37-56
behavior_732163_2024-08-23_13-36-21_processed_2024-08-24_06-57-41

closed 2024-08-26 by alexpiet