AllenNeuralDynamics/aind-fip-dff #10: processed FIP timeseries have double timestamps

alexpiet, 2024-08-15:


alexpiet, 2024-08-26:
Spot checking this issue in different data assets:
- `behavior_705599_2024-06-26_13-04-27.nwb`
- `behavior_700708_2024-05-30_11-16-22.nwb`

alexpiet, 2024-08-27:
@j-friedrich says this is an artifact of pandas combining two preprocessing methods, instead of separating them into two different columns

jeromelecoq, 2024-09-10:
Ahad needs to test stability of this issue.

alexpiet, 2024-09-10:
Test this pull request: https://github.com/AllenNeuralDynamics/aind-fip-dff/pull/7

Ahad-Allen, 2024-09-10:
Will test with a copy of the pipeline using that pr's branch! :)

Ahad-Allen, 2024-09-11:
I think this just needs to be merged, I'll hand it off to ya'll for that! :)

alexpiet, 2024-09-11:
@Ahad-Allen can you point me to the output of the test?

Ahad-Allen, 2024-09-11:
Just shared :)

alexpiet, 2024-09-11:
Confirmed this problem is resolved by the PR.