AllenNeuralDynamics/aind-fip-dff #12: Evaluation of FIP preprocessing

alexpiet, 2024-08-27:


alexpiet, 2024-08-27:
Session with poor baseline subtraction:

behavior_717531_2024-08-23_08-58-49
[image]

alexpiet, 2024-08-27:
Flat signal at the start of the session. Is this because excitation wasn't running? Or is this an artifact of processing?
behavior_711042_2024-08-23_13-27-08

[image]

alexpiet, 2024-08-27:
Session with G_1_preprocessed and G_2_preprocessed all NAN. G_1 and G_2 are NOT all NaN
behavior_708031_2024-08-26_08-49-03

alexpiet, 2024-08-27:
Strange artifacts in FIP signal
behavior_717531_2024-08-26_09-08-42

[image]

jeromelecoq, 2024-09-24:
this need @hagikent eyes. @j-friedrich made PR and applied to real and synthetic data.

jeromelecoq, 2024-10-15:
@hagikent will use the repo from @rachelstephlee to check sessions and start checking these.

alexpiet, 2024-10-15:
repo that contains notebook for looking at data: https://github.com/AllenNeuralDynamics/aind_ssm_hackathon

hagikent, 2024-10-15:
wait for motion correction integration

hagikent, 2024-11-05:
@j-friedrich 
I still see you are pushing commits to the MC PR. Still working on it or should I start testing it?

hagikent, 2024-11-12:
-Tested with a couple representative sessions and MC itself is basically working. I only have a minor suggestion:
https://github.com/AllenNeuralDynamics/aind-physio-arch/issues/265#issuecomment-2470925215

-@j-friedrich (w/ @KasparP and myself) was working on simulation-based novel preprocessing framework that ended up too aggressively subtracting baseline (back then Stefano was actively evaluating this). Any improve since then? Do we completely drop the effort? Adding it as a module in the future update (in addition to the basic poly/exp/bright)?

-I need to learn how to efficiently test this with:
1. more data
2. in code ocean
3. in a manner as close to the actual pipeline as possible 
Can @Ahad-Allen or @rachelstephlee help me on this?

j-friedrich, 2024-11-12:
Hi @hagikent, we haven’t dropped the effort—the processing still relies on simulations as outlined in issue #2. We've moved away from using a running percentile filter for baseline estimation, which is effective for 2P imaging but too flexible for fiber photometry, as it absorbs some neural activity into the baseline. Instead, we're using more rigid parametric models.
We also moved away from using Factor Analysis for estimating motion attenuation, as it can mistakenly interpret neural activity as motion when the green and red channels have correlated neural signals.

As detailed in #2, 14 parameters (highlighted in red in the equations) are used for simulations. Each of these parameters was fit across real datasets (autofluor, dead-sensor, 4NM), with the median value across datasets representing a "typical" case. In the [dFF comparison capsule](https://codeocean.allenneuraldynamics.org/capsule/3443145/tree), simulations vary one parameter at a time while keeping others at their "typical" values. Yes, this "naive" approach ignores parameter correlations but is simple and efficient (akin to Naive Bayes).

To explore the three methods (poly, exp, bright) on simulated data, visit the [capsule](https://codeocean.allenneuraldynamics.org/capsule/3443145/tree) and check the `/scratch` directory. There, you’ll find plots for each fiber and channel in subdirectories, showing the raw and ground truth traces with the three fits. For example:

[image]

Additionally, there are two large summary plots (`/scratch/comparison_dff.pdf` and `/scratch/comparison_dff+mc.pdf`) with 14 rows (one per parameter) and four columns ("RMSE," "bias," "std," and "corr"). If you’re concerned with how well the preprocessed traces match the ground truth, focus on the RMSE column. If bias isn’t an issue (i.e., you don’t mind if traces aren’t centered at 0), the "std" column may be more relevant; if both bias and scale aren’t critical, check the "corr" column.

hagikent, 2024-11-12:
>moved away from using Factor Analysis

Ah, I see. I remember the previous implementation was way more complicated (and longer) and I thought "bright" is something different. Now all clear, thanks!

hagikent, 2024-11-15:
@j-friedrich 
Did you have a chance to check this out? Thanks!
https://github.com/AllenNeuralDynamics/aind-physio-arch/issues/265#issuecomment-2470925215

j-friedrich, 2024-11-15:
Good catch @hagikent. Just fixed it, easy and obvious in hindsight [#265](https://github.com/AllenNeuralDynamics/aind-physio-arch/issues/265#issuecomment-2470925215)

jeromelecoq, 2024-11-19:
Duplicated with #2

closed 2024-11-19 by jeromelecoq