AllenNeuralDynamics/aind-fip-dff #15: Bug: Generated processed NWB files do not contain any NWB, just the output

rachelstephlee, 2024-09-26:


alexpiet, 2024-09-26:
Lots of older assets need to be reprocessed. Anything processed earlier than 09-12 has faulty FIP preprocessing.

rachelstephlee, 2024-09-26:
oops thanks. Yea Ahad let me know.

Ahad-Allen, 2024-09-26:
@rachelstephlee  to add some details but it seems that several assets might be missing their reward label info or have it stored using different keys. Will investigate when I get a chance to see if this affects a specific range or only certain assets

Ahad-Allen, 2024-09-26:
I think the issue is related to keys, but it may be better for someone with more understanding of what the keys and experimental inputs should look like to take a look. I created a debug version of the capsule that may be helpful to work in:
https://codeocean.allenneuraldynamics.org/capsule/9612098/tree 
It looks like    ` earned = getattr(Obj, "B_Earned{}WaterStartTime".format(side), [])
    manual = getattr(Obj, "B_Manual{}WaterStartTime".format(side), [])
    auto = getattr(Obj, "B_Auto{}WaterStartTime".format(side), [])
    rewards = getattr(Obj, "B_{}RewardDeliveryTimeHarp".format(side), [])`
will all fail due to this switch up in keys.

alexpiet, 2024-10-02:
What I think is happening:
EarnedWater, ManualWater, and AutoWater are labels that were added at different times. RewardDeliveryTime will also contain the actual time of reward delivery. Most of the time, most rewards will be earned water

What I think we should:
If AutoWater labels do not exist, assume there were no AutoWater rewards
If ManualWater labels do not exist, assume there were not ManualWater rewards
If EarnedWater labels do not exist, assume all rewards not AutoWater or ManualWater were EarnedWater

arielleleon, 2025-03-31:
@Ahad-Allen will check if this is still an issue

Ahad-Allen, 2025-04-03:
These are rerunning. I tried to run them during office hours yesterday but the vpn was having issues due to internet upgrades at the institute.

Ahad-Allen, 2025-04-04:
These all worked

closed 2025-04-04 by Ahad-Allen