AllenNeuralDynamics/aind-fip-dff #16: All NaNs in processed timeseries

alexpiet, 2024-10-10:
- behavior_700708_2024-06-27_08-43-37_processed_sfn_rachel
- ID: 37a74f48-c2c5-4921-bacd-fcd450126806
- `G_1` and `G_2` have numeric values
- `G_1_dff-*` and `G_2_dff-*` are all NaNs

alexpiet, 2024-10-10:
@Ahad-Allen Can you take a look at this?

jeromelecoq, 2024-10-15:
@j-friedrich will look into this and figure out what is going on.

j-friedrich, 2024-10-30:
`G_1` and `G_2` include NaNs at the end of the recording, which - looking at the traces - should have stopped earlier. Do we always have to cut off a few frames at the end?
[image]

dyf, 2024-11-05:
Detect first NaN, discard all other data from the session for the purpose of dF/F and motion correction.

Log that this happened as a QC metric.

hagikent, 2024-11-05:
TLDR: QA should take care of this by calculating derivative and then tossing the data after.

The best metrics would be:
-sudden large change in the 1st derivative of (multiple, more than two) traces simultaneously

Some rationale for why hardware/rawdata cannot be NaN:
-CMOS minimum values are 255 (in 16it)
-there is always sensor floor noise, so cannot be constatly 255 even there is no photon
-if you unplug patchcables in a really bright situation, the value would be saturated at 655xx (max in 16bit)
-if you unplug in a dark situation, ~255+noise

__
@j-friedrich 
could you please go through processed data and collect sessions with NaN (after preprocessing)?
we can then check them together to confirm if this mostly happens when unplugged before acquisition stopped or not.

closed 2024-11-19 by jeromelecoq