AllenNeuralDynamics/aind-fip-dff #21: Patch for Iso channel column number error

hagikent, 2024-11-22:
The issue is described here.
https://github.com/AllenNeuralDynamics/aind-behavior-blog/issues/735

Briefly, large fraction of Iso channel data.csv files have 9 columns instead of 6 (timestamp, ROI0, 1, 2, 3, Blank). Practically, this is not an issue as what happens is having three additional columns with all 0. 
[image]

This issue has been caused by an bug in the following Bonsai code (will fix this shortly):
[image]

As this is confusing (why 9 column? why different from other two channels?), we should take care of this before nwb packaging.

jeromelecoq, 2024-12-03:
Plan to fix early January.

rachelstephlee, 2024-12-10:
This is fixed on the experimenter rig. 

When we will be processing older data, we will fix the pipeline side as well. 
- Will need to eliminate some 0-filled data.

closed 2024-12-10 by rachelstephlee