AllenNeuralDynamics/aind-fip-dff #23: ValueError: Axis limits cannot be NaN or Inf

alexpiet, 2024-12-04:
This is causing sessions to crash

https://codeocean.allenneuraldynamics.org/data-assets/7f5ad1a6-3b54-441d-925d-e73f13c4a514/behavior_754280_2024-12-02_11-09-44_processed_2024-12-03_07-16-54/output?fullScreen=true&hideDetails=true

Ahad-Allen, 2024-12-04:
Looking at 757212, it does appear to have an empty line present at the end of the CSV files for red iso and green that is not present in mice that have not thrown this NaN error.

Ahad-Allen, 2024-12-04:
2c67fd77-bd6e-40a4-a100-418b4c9c9d2e has 102778 values for rising times in its harp json file but 104430 values in its CSV file for fiber data. Because of this, nan values get appended for each timestamp value after the 102778th in the data frames we eventually add to the NWB file. I am not sure why there are more fiber than harp values in the raw data. Perhaps @alexpiet  or @hagikent have some insight on that?

alexpiet, 2024-12-04:
That seems like an alarming data issue, I don't have any insight into why that would be the case.

hagikent, 2024-12-05:
thanks both for the report. 

>alarming data issue
Indeed, yes... I thought instability of HARP sync-pulses (rising times) recording issue had been solved...

So far I don't know what exactly is messed up here. I need to see the raw data.

In the past, Stefano and I implemented an interpolation-based way to recover missed rising signals, which will solve most minor drop cases but if there are extreme cases, it could penetrate.  
To figure out what's going on, I need to focus on this, which is a bit difficult for now (attending a grant meetup in CA until Sun). I will work on it next week.

j-friedrich, 2024-12-07:
The plotting issue that caused the ValueError of this issue is fixed in my recent push to aind-fip-dff a few minutes ago. The pipeline should now run through even if some timestamps are NaNs. However, it currently does not correct them but retains them as NaNs.

j-friedrich, 2025-01-28:
Closing this ticket as the sync pulse error mentioned in the discussion is being tracked in a separate ticket, and the plotting error referenced in the title has already been resolved.

closed 2025-01-28 by j-friedrich