AllenNeuralDynamics/aind-fip-dff #32: Sync Pulse error recovery

hagikent, 2025-01-07:
Harp is recording 20Hz sync pulse (TTL pusle trains from FIP stream) as a Digital-Input.
This recording is somewhat fragile and missing some pulses infrequently.

As the missed pulses can be detected by calculating diff between pulses, preprocessing is taking care of this.
However, there seems to be edge cases that cannot be recovered by the current preprocessing.

@j-friedrich will check/evaluate the algorithm and work on improvement.

Now QCcapsule (soon-to-be a part of pipeline) is detecting this issue automatically, we will have better understanding of how frequently this is happening.

hagikent, 2025-01-07:
forked from: https://github.com/AllenNeuralDynamics/aind-fip-dff/issues/27

j-friedrich, 2025-01-10:
This asset is currently attached to [aind-fip-qc-raw](https://codeocean.allenneuraldynamics.org/capsule/5044779/tree) which doesn't show any errors in SyncPulse.

hagikent, 2025-01-10:
@j-friedrich 
My bad, one asset off! (was attaching multiple assets and counting the loop number from 1 not 0)
behavior_754430_2024-12-16_13-03-21

Until QC runs on the pipeline, please look for several recent FIP sessions with `behavior_` and check if some gives you the error.

JeremiahYCohen, 2025-01-14:
@hagikent  shared data, @j-friedrich  trying to understand temporal gap that is not a multiple of 15 ms. Follow up with Teams chat with @bruno-f-cruz

Ahad-Allen, 2025-02-11:
@hagikent to schedule a meeting this week/next week.

hagikent, 2025-02-19:
We met Bruno and learned how to investigate the source of the pulse recording errors.
@j-friedrich will do this investigation with several error-causing sessions, which are easy to find find with the current QC.

j-friedrich, 2025-03-11:
See [#75](https://github.com/AllenNeuralDynamics/aind-fip-nwb-base-capsule/issues/75) and [#42](https://github.com/AllenNeuralDynamics/aind-fip-dff/issues/42#issuecomment-2711953105).

It seems like there’s an issue with the JSON generation within the Bonsai code.
Repeating myself, when Kenta and I spoke with Bruno, he pointed out that the timestamps in the behavior JSON are written not only after passing through the entire Bonsai code but also after being sent to the Python GUI — leaving room for multiple potential failure points. He suggested that it would be better to track closer to the source, at the beginning of the Bonsai code, to minimize such issues.
Additionally, Bruno recommended tracking the camera metadata (ReferenceTime, CameraFrameNumber, and CameraFrameTime) as in [behavior_videos.md](https://github.com/AllenNeuralDynamics/aind-file-standards/blob/main/file_formats/behavior_videos.md). He mentioned that he already flagged this with you and Xinxin. While this might not directly solve the current issue, having that info could help trace the root cause.

hagikent, 2025-03-11:
For this specific case (bowen's session), yes, the issue seems to be stemming from Bonsai stopped in the middle of the sessions. However, this is not tracking the general issue down at all. 

We observe infrequent but non-rare TTL **single-drops** and **double-countings**. 
Are you 100% sure that they are also json packaging issue? If so, why?

Could you please analyze several sessions with mixed TTL issues by comparing timestamps in json and in harp-messages (this is what Bruno suggested and we agreed on) to conclude if the issue is either hardware-level or software-level? And if software level, within Bonsai stream or in OSC communication between Bonsai-Python GUI?

@j-friedrich 
We need those diagnosis, thanks!

dougollerenshaw, 2025-03-26:
@j-friedrich, @hagikent pinged me about this issue offline. Sounds like it's high priority for him and blocking some followup work. Could you update on progress when you're back in the office?

hagikent, 2025-04-04:
@j-friedrich any update on this? I see this is the highest priority / long-standing issue that has been waiting for your action.
Was the agreement with Bruno/me during the discussion on Feb (summarized above) not clear/actionable enough? 
If so, how about:
1.picking 20 sessions with `SyncPulseQC`=Fail
2.diagnosing what is going wrong
3.checking if the implemented recovery routine properly fixes identified issues (or not)
4.if not, update the recovery process

Let me know how others including myself can be helpful to solve this faster! Thanks!

for project/progress visibility: @dougollerenshaw @dyf @arielleleon @alexpiet @JeremiahYCohen

j-friedrich, 2025-04-05:
Oh, I’m just waiting for the folks responsible for **raw** data to get their shit together (excuse the American English) to fix this **raw** data issue raised by **raw**-qc, instead of relying on **processing** folks to clean up their mess (as Stefano did out of the goodness of his heart). Until then, this will be an open issue, so give it a few months, hopefully less now that Bruno actually is on it himself. 
Well, actually closing cause it’s in the wrong repo, aind-fip-dff has causally noting to do with **raw** data, and reopening in the proper place [#75](https://github.com/orgs/AllenNeuralDynamics/projects/75/views/1?filterQuery=platform%3Afib&pane=issue&itemId=105129014&issue=AllenNeuralDynamics%7CFIP_DAQ_Control%7C4).

closed 2025-04-05 by j-friedrich