AllenNeuralDynamics/aind-fip-dff #34: make the repo pip installable

hagikent, 2025-02-11:
The preprocessing functions in this repo is useful in many non-pipeline-based anlysis including local analysis and pilot data exploration. To make the preprocessing consistent across researchers in house, this repo/capsule should be maintained to be pip-installable.

Seemingly @ZhixiaoSu did it for her own sake by making a private branch.
Could @j-friedrich update the main to do the same? Sue mentioned this is a code-ocean source template issue.

alexpiet, 2025-02-11:
What I've done in other capsules is have the code live in a pip installable repo, and then in a separate capsule call the functions from that pip-library. Its a little annoying because you have to maintain two repos, but I think its the only way to have a clean CO capsule with a installable library. Open to alternative suggestions

ZhixiaoSu, 2025-02-11:
I agree with @alexpiet, it also makes it easier to reuse the funcitons in analysis CO capsules. Capsules can manage pip -e packages much more reasonably than git cloned repo. (I learned in a hard way.) Here's the branch that is pip installable: https://github.com/AllenNeuralDynamics/aind-fip-dff/tree/local_copy_for_LC-NE_manuscript
The changes were: add pyproject.toml (edit accordingly); add setup.py; put all code under src and add __init__.py.

hagikent, 2025-02-12:
@j-friedrich Could you work on this and maintain as Alex/Sue suggest?

alexpiet, 2025-11-06:
Bumping this for triaging

j-friedrich, 2025-11-07:
Not for me. Maybe someone else has already figured out what needs to be added to a capsule so that people who don’t want to simply run the Code Ocean capsule can just do:
```
pip install git+https://github.com/AllenNeuralDynamics/aind-fip-dff
```

dougollerenshaw, 2025-11-11:
@j-friedrich Who should this be assigned to?

j-friedrich, 2025-11-24:
I noticed Tom has been experimenting with turning a capsule into a library, so I took some inspiration from that and created a new branch that transitions the repo into a proper Python package.

You can now install it with:  
`pip install git+https://github.com/AllenNeuralDynamics/aind-fip-dff@pip`
and import it with:  
`import aind_fip_dff`. 
The functions that are defined in `run_capsule.py` are available directly at the package root, while additional helper functions live in `aind_fip_dff.utils`.

To run the main `run_capsule.py` script, use:  
`python -m aind_fip_dff`
(you can also append command-line arguments handled via argparse).

closed 2025-11-24 by j-friedrich

hagikent, 2025-11-24:
Thanks @j-friedrich ! This is great!