AllenNeuralDynamics/aind-fip-dff #36: Processing QC only shows Fiber1 and Fiber2

hagikent, 2025-02-12:
All 4 functional ROIs per channel should be evaluated and visualized, although ROI3/4 are rarely used at the moment.
Also, for consistency with raw-qc, ROI0, ROI1, ROI2, ROI3 naming convention would be better.

j-friedrich, 2025-02-27:
Choose Kenta's naming scheme for the QC plots as part of PR #38. (Though there is inconsistency with the NWB where counting starts at 1).

The number of ROIs is not an issue of this `aind-fip-dff` caspule, but the `aind-fip-nwb-base-capsule` where the number is hard-coded to 2, [here](https://github.com/AllenNeuralDynamics/aind-fip-nwb-base-capsule/blob/d23ec9350016b9d3433d4a4bf3581870d8e079f0/code/util/aligned_fiber_to_nwb.py#L695).

If changed to 4, the downstream `aind-fip-dff` caspule indeed processes and visualizes all ROIs
Should we just do that?

hagikent, 2025-02-28:
>Though there is inconsistency with the NWB where counting starts at 1

True. What do you think is the best to make everything consistent? At the experiments level, 0-starting (ROI0, ROI1, ROI2...) is sort of inevitable.

>aind-fip-nwb-base-capsule where the number is hard-coded to 2

ah, nice catch... then this has to be fixed.

j-friedrich, 2025-02-28:
happy to change the hard-coded 2 to a 4, if we always have (at least) 4 and it's ok to hard-code the value

I don't know whether there is a NWB rule to start counting at 1, if not we could just rename those

hagikent, 2025-02-28:
>if we always have (at least) 4

yes.

dougollerenshaw, 2025-03-20:
@j-friedrich and @hagikent looks like there was active work here but no comments in the last 3 weeks. Is this still moving?

j-friedrich, 2025-04-01:
This is already part of the new(est) released pipeline, thus closing.

closed 2025-04-01 by j-friedrich