AllenNeuralDynamics/aind-fip-dff #37: "bright" method seems malfunctioning in some sessions

hagikent, 2025-02-14:
e.g.
https://qc.allenneuraldynamics.org/qc_app?id=247cd3ae-37f5-4bd3-af8c-ec8dbcc5cfa2&active_evaluation=7
[image]

@j-friedrich could you please investigate what's going on?

@ZhixiaoSu have you by chance experienced something similar or other weird behavior of bright in your local analysis?

ZhixiaoSu, 2025-02-14:
No, I have never seen a very off bright fit like this.

j-friedrich, 2025-02-14:
I already noticed that some of the default parameters can in rare cases lead to failure and am updating those [#491](https://github.com/AllenNeuralDynamics/aind-physio-arch/issues/491).  
Thanks for pointing me to that dataset, I'll check whether it will work with my new defaults

hagikent, 2025-02-19:
another malfunctioning session:
https://qc.allenneuraldynamics.org/qc_app?id=f83dde1d-b478-41cd-b932-06c61f0debf9&active_evaluation=0

j-friedrich, 2025-02-28:
I verified that the failed datasets run fine with the updates in [PR #38](https://github.com/AllenNeuralDynamics/aind-fip-dff/pull/38)

closed 2025-03-11 by j-friedrich

rachelstephlee, 2025-03-18:
also found in session '746346_2025-01-22' for G_1

j-friedrich, 2025-04-01:
I just verified that session '746346_2025-01-22' works ok with the recent release of the pipeline. Just rerun it @rachelstephlee

Ahad-Allen, 2025-04-02:
Will reprocess today