AllenNeuralDynamics/aind-fip-dff #39: take care of "multiple recordings in a session" edge case

hagikent, 2025-02-22:
This following session crashed QC capsule. I guess base and preprocessing capsules need to be updated too?

[image]

Note, this would happen when experimenter start/stops FIP-Bonsai multiple times.

j-friedrich, 2025-04-02:
This data asset was apparently reprocessed fine. ID: [08bf2d69-6c15-4fe3-9f61-9b1020d98487](https://codeocean.allenneuraldynamics.org/data-assets/08bf2d69-6c15-4fe3-9f61-9b1020d98487/behavior_764702_2025-02-11_12-13-17_processed_2025-03-12_00-22-53?filters=N4IgjgrgpgTgniAXCADADgEYDMBMATANgE4BaAgYwEYBWEgFiygGYSisDLWNKUcU8iaOmgDsIADQgAzgHsYAFyQhyMKAEN5UPBJB4oU8knkxokgA5qA5lCSVzsAApWbiPpLV4AbmoB25LUjAAL6SALYQADbyAJYAYtFRsFKBQUFAA)

Here the starting and stopping yielded some empty (zero byte) .csv and .bin files. Are there other scenarios with multiple legit recordings in a session, or can we close @hagikent?

hagikent, 2025-04-02:
@j-friedrich 
In the QC, we decided to target the **longest data** as >99% time, others are garbage. (i.e. it's very rare that a session folder contains meaningful two datasets)

Could you do the same in `dff` and `base` while logging an error message to grafana "multiple files found".

c.f. https://github.com/AllenNeuralDynamics/aind-fip-qc-raw/pull/17

j-friedrich, 2025-04-02:
Sounds good.
Note to myself: Fix this in function `detect_fiber_photometry_system`

j-friedrich, 2025-04-09:
This is already fixed as part of above PR within _aind-fip-nwb-base-capsule_ and tracked as [#79](https://github.com/orgs/AllenNeuralDynamics/projects/75/views/1?filterQuery=multi&pane=issue&itemId=98976659&issue=AllenNeuralDynamics%7Caind-fip-nwb-base-capsule%7C79), thus closing this ticket.

closed 2025-04-09 by j-friedrich