AllenNeuralDynamics/aind-fip-dff #40: ValueError: The length of the input vector x must be greater than padlen, which is 9.

alexpiet, 2025-02-24:
https://codeocean.allenneuraldynamics.org/data-assets/55e3208c-f0a1-4333-9d5a-3f3873c85401/behavior_711039_2024-08-23_10-02-16_processed_2025-02-22_04-37-09/output?fullScreen=true&hideDetails=true

```
 + python -u run_capsule.py
  /tmp/nxf.8rz1F5EgZK/capsule/code/utils/preprocess.py:352: UserWarning: kernel_size exceeds volume extent: the volume will be zero-padded.
    tc_filtered = medfilt(tc_cropped, kernel_size=kernel_size)
  Traceback (most recent call last):
    File "/tmp/nxf.8rz1F5EgZK/capsule/code/run_capsule.py", line 314, in <module>
      df_fip_pp_nwb, df_PP_params, df_fip_mc = batch_processing(
    File "/tmp/nxf.8rz1F5EgZK/capsule/code/utils/preprocess.py", line 470, in batch_processing
      NM_preprocessed, NM_fitting_params = chunk_processing(
    File "/tmp/nxf.8rz1F5EgZK/capsule/code/utils/preprocess.py", line 353, in chunk_processing
      tc_filtered = tc_lowcut(tc_filtered, sampling_rate)
    File "/tmp/nxf.8rz1F5EgZK/capsule/code/utils/preprocess.py", line 27, in tc_lowcut
      return sosfiltfilt(sos, tc)
    File "/opt/conda/lib/python3.10/site-packages/scipy/signal/_signaltools.py", line 4810, in sosfiltfilt
      edge, ext = _validate_pad(padtype, padlen, x, axis,
    File "/opt/conda/lib/python3.10/site-packages/scipy/signal/_signaltools.py", line 4574, in _validate_pad
      raise ValueError("The length of the input vector x must be greater "
  ValueError: The length of the input vector x must be greater than padlen, which is 9.
```

dougollerenshaw, 2025-03-20:
Talking to Alex. He's unsure of the cause. @j-friedrich , Alex thinks you're best equipped to solve. Can you take this on? Assigning to you for now.

j-friedrich, 2025-04-09:
Just seeing this now, due to my vacation. Hm, the `behavior.json` includes 95570 time stamps, but the fiber .csv files only 25 (`FIP_DataR_2024-08-23T10_02_20.csv`), i.e. merely 1.25 seconds of fiber recording. Should we turn this into a behavior-only session, @alexpiet?

Currently we check in [aind-fip-nwb-base-capsule](https://github.com/AllenNeuralDynamics/aind-fip-nwb-base-capsule/blob/e801304b9243ef1c99ca7b8b6573e5118d3bea0a/code/run_capsule.py#L234) whether fiber data exists or it's behavior-only, could add a check that if it exists, it must be of reasonable (couple of minutes??) lengths. 

(However, this will need to be updated again once there are separate fiber and behavior pipelines.)

alexpiet, 2025-04-09:
Seems reasonable to me to ensure a minimum length of the fiber data. @hagikent what is a good threshold?

hagikent, 2025-04-09:
Double-checked the rawdata asset of this session, and I confirm that indeed FIP recording seems to be failed (likely ThorFIP.bonsai stopped) immediately.

Let's say 5min for now? In raw-qc, I set the metrics as 15min but when combined with other behavior, a proper session may be shorter than 15min.

j-friedrich, 2025-04-10:
Ok, added a 5min threshold as condition for appending the FIP data to my PR [#90](https://github.com/AllenNeuralDynamics/aind-fip-nwb-base-capsule/pull/90). 

Might still change it from "if FIP is less than 5 min than its behavior-only" to "if FIP is less than 5min **but behavior is much longer** than its behavior-only"

j-friedrich, 2025-04-11:
Addressed by PR [#90](https://github.com/AllenNeuralDynamics/aind-fip-nwb-base-capsule/pull/90), thus assigning to @Ahad-Allen for the next merge, sync, and release cycle.

Ahad-Allen, 2025-04-17:
Merged and deployed

closed 2025-04-17 by Ahad-Allen