AllenNeuralDynamics/aind-fip-dff #41: DataDescription validation errors

alexpiet, 2025-02-24:
https://codeocean.allenneuraldynamics.org/data-assets/5d0c649c-960f-4e34-8eba-33a8dd1d063e/behavior_730942_2024-09-20_17-19-20_processed_2025-02-23_01-19-35/output?fullScreen=true&hideDetails=true

```
  Traceback (most recent call last):
    File "/tmp/nxf.HozyjJ8InF/capsule/code/run_capsule.py", line 388, in <module>
      write_output_metadata(
    File "/tmp/nxf.HozyjJ8InF/capsule/code/run_capsule.py", line 113, in write_output_metadata
      new_dd = dd_upgrader.upgrade()
    File "/opt/conda/lib/python3.10/site-packages/aind_metadata_upgrader/data_description_upgrade.py", line 313, in upgrade
      return construct_new_model(data_desc_dict, self.model_class, self.allow_validation_errors)
    File "/opt/conda/lib/python3.10/site-packages/aind_metadata_upgrader/utils.py", line 24, in construct_new_model
      raise e
    File "/opt/conda/lib/python3.10/site-packages/aind_metadata_upgrader/utils.py", line 14, in construct_new_model
      return model_type.model_validate(model_inputs)
    File "/opt/conda/lib/python3.10/site-packages/pydantic/main.py", line 568, in model_validate
      return cls.__pydantic_validator__.validate_python(
  pydantic_core._pydantic_core.ValidationError: 2 validation errors for DataDescription
  funding_source
    List should have at least 1 item after validation, not 0 [type=too_short, input_value=[], input_type=list]
      For further information visit https://errors.pydantic.dev/2.8/v/too_short
  investigators
    List should have at least 1 item after validation, not 0 [type=too_short, input_value=[], input_type=list]
      For further information visit https://errors.pydantic.dev/2.8/v/too_short
```

alexpiet, 2025-03-13:
@dyf @jtyoung84 These assets failed to process because the `data_description.json` failed to validate. All of these assets failed for the same reason, an empty list for `funding_source` and `investigators`. These data_description.json files were from a timepoint when the acquisition code was creating the data_description.json, instead of letting the transfer service perform it. 

Is it possible to create a new data_description.json from the code the transfer service runs? 

status | date | mouse | acquisition rig | asset_name | ERROR
-- | -- | -- | -- | -- | --
error processing data | 2024-09-20 | 730942 | W10DT714030 | behavior_730942_2024-09-20_17-19-20 | https://github.com/AllenNeuralDynamics/aind-fip-dff/issues/41
error processing data | 2024-09-25 | 728570 | W10DT714027 | behavior_728570_2024-09-25_16-34-40 | https://github.com/AllenNeuralDynamics/aind-fip-dff/issues/41
error processing data | 2024-09-26 | 741210 | W10DT714027 | behavior_741210_2024-09-26_17-59-33 | https://github.com/AllenNeuralDynamics/aind-fip-dff/issues/41
error processing data | 2024-09-26 | 743794 | W10DT714030 | behavior_743794_2024-09-26_18-06-49 | https://github.com/AllenNeuralDynamics/aind-fip-dff/issues/41
error processing data | 2024-09-26 | 747315 | W10DT714084 | behavior_747315_2024-09-26_17-56-18 | https://github.com/AllenNeuralDynamics/aind-fip-dff/issues/41
error processing data | 2024-09-28 | 747315 | W10DT714084 | behavior_747315_2024-09-28_17-53-22 | https://github.com/AllenNeuralDynamics/aind-fip-dff/issues/41
error processing data | 2024-09-30 | 741210 | W10DT714027 | behavior_741210_2024-09-30_18-04-39 | https://github.com/AllenNeuralDynamics/aind-fip-dff/issues/41
error processing data | 2024-09-30 | 747313 | W10DT714084 | behavior_747313_2024-09-30_18-01-53 | https://github.com/AllenNeuralDynamics/aind-fip-dff/issues/41

jtyoung84, 2025-03-13:
Since it's only 8 data sets, it might be easier to pull the data_description information from DocumentDB, manually add the missing information, and then upload the changes back to DocumentDB. The indexer will replace the data_description.json files when it runs (every four hours).

dougollerenshaw, 2025-03-20:
Talking to @alexpiet. Steps to follow here:
1) I will make a data migration script to update the data descriptions in docdb (self assigning for now)
2) Then I will assign to @Ahad-Allen , who will then trigger processing on these assets.

It looks like it's failing to validate due to missing funding info. I will investigate.

alexpiet, 2025-10-27:
Thanks for looking into this @dougollerenshaw. That seems correct to me. I would just double check with @bwtan-allen that the 61 sessions are all from his mice, and thats the correct project code/funding/project name.

dougollerenshaw, 2025-10-27:
This should be a full table of all 61 assets:

| _id | name | creation_time | project_name | modality |
|-----|------|---------------|--------------|----------|
| 9ea01e9c-561f-4cd0-aa30-d02ce5130074 | behavior_730942_2024-09-20_17-19-20 | 2024-09-20T17:19:20 | Genetic Perturbation Platform | behavior, behavior-videos |
| 6f2a7a91-6139-4507-abbc-34b8a1b752e3 | behavior_730945_2024-09-18_17-24-10 | 2024-09-18T17:24:10 | Genetic Perturbation Platform | behavior, behavior-videos |
| 5c2c5e1b-f590-4ee4-85ed-f7624d130653 | behavior_730945_2024-09-19_17-26-54 | 2024-09-19T17:26:54 | Genetic Perturbation Platform | behavior, behavior-videos |
| 5e1889b4-be77-476c-a115-3cacb192c0ba | behavior_730945_2024-09-20_17-12:16 | 2024-09-20T17:12:16 | Genetic Perturbation Platform | behavior, behavior-videos |
| bcc2e7db-2e65-4387-808a-b0b03888cd31 | behavior_730942_2024-09-18_17-31-41 | 2024-09-18T17:31:41 | Genetic Perturbation Platform | behavior, behavior-videos |
| 80f8c1d0-63e9-4a8c-9ba0-e41ceb957566 | behavior_730942_2024-09-19_17-35-50 | 2024-09-19T17:35:50 | Genetic Perturbation Platform | behavior, behavior-videos |
| 251e6b3e-f86a-47e6-9a21-3097ee6e9b03 | behavior_728568_2024-09-25_16-38-33 | 2024-09-25T16:38:33 | Genetic Perturbation Platform | behavior, behavior-videos |
| 6e6dd6b3-7104-45e2-933d-53cf4d58cf50 | behavior_728568_2024-09-23_18-32-08 | 2024-09-23T18:32:08 | Genetic Perturbation Platform | behavior, behavior-videos |
| a2591ac9-dbd2-43b0-aab3-0ee395f712c9 | behavior_728568_2024-09-24_17-38-24 | 2024-09-24T17:38:24 | Genetic Perturbation Platform | behavior, behavior-videos |
| 1241c4e7-e4b3-41a9-8657-360b0fc695c6 | behavior_728568_2024-09-26_18-12-39 | 2024-09-26T18:12:39 | Genetic Perturbation Platform | behavior, behavior-videos |
| 99ad1192-937d-4aaf-b1e2-5e339b688bde | behavior_728568_2024-09-27_17-32-07 | 2024-09-27T17:32:07 | Genetic Perturbation Platform | behavior, behavior-videos |
| 387702f6-435e-4180-83fb-1d8fb8bd763f | behavior_728568_2024-09-28_18-02-01 | 2024-09-28T18:02:01 | Genetic Perturbation Platform | behavior, behavior-videos |
| 64dfbc73-6ef7-4557-8522-4428430c3405 | behavior_728568_2024-09-29_17-06-22 | 2024-09-29T17:06:22 | Genetic Perturbation Platform | behavior, behavior-videos |
| 389f9121-e8d5-486e-b1a8-6488757b91cf | behavior_728568_2024-09-30_18-10-19 | 2024-09-30T18:10:19 | Genetic Perturbation Platform | behavior, behavior-videos |
| 93f4fa6a-d0ed-43f2-9c22-782ab59b7c10 | behavior_728568_2024-10-01_18-11-28 | 2024-10-01T18:11:28 | Genetic Perturbation Platform | behavior, behavior-videos |
| fa05a550-4004-4a25-ac95-fef540118d51 | behavior_728570_2024-09-25_16-34-40 | 2024-09-25T16:34:40 | Genetic Perturbation Platform | behavior, behavior-videos |
| 4c1930eb-6f32-4252-9487-ca7cef418ae1 | behavior_728570_2024-09-23_18-28-44 | 2024-09-23T18:28:44 | Genetic Perturbation Platform | behavior, behavior-videos |
| 39adec21-eb0e-445d-bfd0-0ec3c3ebda85 | behavior_728570_2024-09-24_17-34-38 | 2024-09-24T17:34:38 | Genetic Perturbation Platform | behavior, behavior-videos |
| 8b9e9537-71b5-4b72-a217-ceffb91785c2 | behavior_728570_2024-09-26_18-03-05 | 2024-09-26T18:03:05 | Genetic Perturbation Platform | behavior, behavior-videos |
| 365e40f7-a228-4987-a452-08ae19d680bd | behavior_728570_2024-09-27_17-27-56 | 2024-09-27T17:27:56 | Genetic Perturbation Platform | behavior, behavior-videos |
| 075aceee-73ef-4f0a-9f76-b44de1e004b1 | behavior_728570_2024-09-28_17-58-09 | 2024-09-28T17:58:09 | Genetic Perturbation Platform | behavior, behavior-videos |
| 7698040a-4edd-44f5-b2aa-d87fdce3cdcf | behavior_728570_2024-09-29_17-01-15 | 2024-09-29T17:01:15 | Genetic Perturbation Platform | behavior, behavior-videos |
| 62fed28c-ef28-4670-901d-66d793f6d534 | behavior_728570_2024-09-30_18-06-09 | 2024-09-30T18:06:09 | Genetic Perturbation Platform | behavior, behavior-videos |
| 3e2b0dc9-970d-4b86-980c-da8c72ce0b7c | behavior_728570_2024-10-01_18-06-48 | 2024-10-01T18:06:48 | Genetic Perturbation Platform | behavior, behavior-videos |
| 0fa32b4d-27aa-4e9d-a7a6-7cf836c2e94b | behavior_741211_2024-09-18_17-30-04 | 2024-09-18T17:30:04 | Genetic Perturbation Platform | behavior, behavior-videos |
| bf3c62c6-36e1-40c8-8f75-0cb8117ed6eb | behavior_741211_2024-09-19_17-32-52 | 2024-09-19T17:32:52 | Genetic Perturbation Platform | behavior, behavior-videos |
| 5a1b8a5b-435f-4152-951b-4f323b957d72 | behavior_741211_2024-09-20_17-17-37 | 2024-09-20T17:17:37 | Genetic Perturbation Platform | behavior, behavior-videos |
| 761b06ee-81c6-412f-a2ee-892a8ae1733e | behavior_741210_2024-09-23_18-27-05 | 2024-09-23T18:27:05 | Genetic Perturbation Platform | behavior, behavior-videos |
| 26e7b442-b9de-4e22-b332-981c2cb3ffb7 | behavior_741210_2024-09-24_17-33-15 | 2024-09-24T17:33:15 | Genetic Perturbation Platform | behavior, behavior-videos |
| 77ac9017-95e0-427c-b5c4-d31e6b1ab084 | behavior_741210_2024-09-25_16-32-53 | 2024-09-25T16:32:53 | Genetic Perturbation Platform | behavior, behavior-videos |
| b2f0d127-fcda-4f6b-b938-e7a9b0d650b5 | behavior_741210_2024-09-26_17-59-33 | 2024-09-26T17:59:33 | Genetic Perturbation Platform | behavior, behavior-videos |
| 01b68411-d666-4e76-ae91-a0851b8c66f6 | behavior_741210_2024-09-27_17-26-35 | 2024-09-27T17:26:35 | Genetic Perturbation Platform | behavior, behavior-videos |
| a516e310-13cb-4e6d-bc75-1f9ffebe637e | behavior_741210_2024-09-28_17-56-57 | 2024-09-28T17:56:57 | Genetic Perturbation Platform | behavior, behavior-videos |
[... cut ...]
| e02caedb-98e8-4150-b327-cacfa22bf63e | behavior_747315_2024-09-28_17-53-22 | 2024-09-28T17:53:22 | Genetic Perturbation Platform | behavior, behavior-videos |
| b9aa1b90-8228-49a5-a11c-58d64fe2764a | behavior_747315_2024-09-29_16-51-36 | 2024-09-29T16:51:36 | Genetic Perturbation Platform | behavior, behavior-videos |
| 114e5d93-8795-42c6-a31e-e07674dd7837 | behavior_747315_2024-09-30_17-59-49 | 2024-09-30T17:59-49 | Genetic Perturbation Platform | behavior, behavior-videos |
| 57a76b94-a491-459f-8d86-acb39179ed35 | behavior_747315_2024-10-01_17-57-59 | 2024-10-01T17:57:59 | Genetic Perturbation Platform | behavior, behavior-videos |