AllenNeuralDynamics/aind-fip-dff #42: Time alignment problem in early-stop sessions

bwtan-allen, 2025-03-10:
I have recently noticed that a few sessions stopped early showed misaligned events, an example as follows:

[image]

While the same mouse, same fiber from the rest sessions that are completed to the end all showed responses right at time 0:

[image]

See the affected data assets below:

behavior_763704_2025-03-07_18-45
behavior_769253_2025-03-09_16-17-01.nwb
behavior_769253_2025-03-09_16-56-58.nwb

alexpiet, 2025-03-10:
@hagikent @j-friedrich any ideas? Do you think this is an issue with the fip-dff capsule, or something in data acquisition?

j-friedrich, 2025-03-11:
The fip-dff capsule doesn't perform any alignment — that's handled in the aind-fip-nwb-base-capsule. Alternatively, the issue might already arise during data acquisition if the timestamps in the behavior JSON are off (although, wouldn't that affect both the behavior and fiber data equally?).

When Kenta and I spoke with Bruno, he pointed out that the timestamps in the behavior JSON are written not only after passing through the entire Bonsai code but also after being sent to the Python GUI — leaving room for multiple potential failure points. He suggested that it would be better to track closer to the source, at the beginning of the Bonsai code, to minimize such issues.
Additionally, Bruno recommended tracking the camera metadata (ReferenceTime, CameraFrameNumber, and CameraFrameTime) as in [behavior_videos.md](https://github.com/AllenNeuralDynamics/aind-file-standards/blob/main/file_formats/behavior_videos.md). He mentioned that he already flagged this with you and Xinxin. While this might not directly solve the current issue, having that info could help trace the root cause.

I can take a closer look at the aind-fip-nwb-base-capsule during Thursday's hackathon to see if the issue originates there. However, if the problem occurs earlier, it might require you or Xinxin to revisit the Bonsai code.