AllenNeuralDynamics/aind-fip-dff #43: 1 validation error for DataDescription - investigators

alexpiet, 2025-03-27:
We have been getting this errors 

```
  Validation error in <class 'pydantic._internal._model_construction.ModelMetaclass'>: 1 validation error for DataDescription
  investigators
    List should have at least 1 item after validation, not 0 [type=too_short, input_value=[], input_type=list]
      For further information visit https://errors.pydantic.dev/2.8/v/too_short
  allow validation errors: False
  Stopping
```

The DataDescription file is generated by the data-transfer service, the validation error is because information is missing
(1) Why doesn't the data-transfer service validate the pydantic model before attaching it to the raw data asset? @jtyoung84 @dyf 
(2) Why are we validating the pydantic model in this repo? @Ahad-Allen @j-friedrich 
(3) Why are there projects that are on the approved project list that don't have all the information they require? @saskiad

dougollerenshaw, 2025-03-27:
tagging @dbirman in case he has insight here.

alexpiet, 2025-03-27:
@Ahad-Allen Said on teams:
"we update the data description in the dff capsule to reflect the running of this pipeline"

saskiad, 2025-03-27:
Sheet is fixed. Looks like someone edited and only 3 people have edit access, so I will hunt them down.

dbirman, 2025-03-27:
> (1) Why doesn't the data-transfer service validate the pydantic model before attaching it to the raw data asset? [@jtyoung84](https://github.com/jtyoung84) [@dyf](https://github.com/dyf) OPEN (2) Why are we validating the pydantic model in this repo? [@Ahad-Allen](https://github.com/Ahad-Allen) [@j-friedrich](https://github.com/j-friedrich) RESOLVED: we modify the data_description based on processing in the pipeline (3) Why are there projects that are on the approved project list that don't have all the information they require? RESOLVED: It was a mistake.

I can answer #1 -- we don't validate because we don't enforce valid metadata at the moment. We'll start enforcing at some point after 2.0 launches (with plenty of warning for users). One of the changes is that we're going to ask users to capture all of their own metadata rather than hide that within the data-transfer-service and cause the kind of error you ran into here to be invisible.

alexpiet, 2025-03-27:
Thanks for the explanation @dbirman 

Closing this issue as resolved

closed 2025-03-27 by alexpiet