AllenNeuralDynamics/aind-fip-dff #44: G0-4 vs G1-4 naming conventions in the processed assets

bwtan-allen, 2025-03-31:
Hi folks,

We had a naming convention change in our recent FIP processed data assets. This means the datasets before Feb still keep the original naming conventions with G/R/Iso-1-4, while the ones after Feb are using the G/R/Iso0-4 series. I'm just wondering whether we have plans to unify these inconsistent naming in the past datasets. For now I think individual users can deal with this on their own end.

I'm posting this ticket here per Alex's suggestion @alexpiet. 

Thanks,
Bowen

alexpiet, 2025-03-31:
@dougollerenshaw We had been using names like Fibers 1,2,3,4. Now they are Fibers 0,1,2,3.

I believe this was an intentional change by @j-friedrich and @hagikent but I can't find the PR or documentation.

hagikent, 2025-03-31:
See this earlier issue:
https://github.com/AllenNeuralDynamics/aind-fip-dff/issues/36

Here I suggested changing the two visualizations consistent but seemingly different channels are wrongly packaged into nwb. I raised this >2w to @j-friedrich but no feedback yet. 

___
And, @dougollerenshaw and I are in the middle of discussing what would be the best to reduce inconsistency across the platform. 0 start vs 1 start inconsistensy is everywhere and this is just the tip of iceberg. We should solve this in an organised way instead of by a quick patch.
Disscussion can be found here: https://github.com/AllenNeuralDynamics/aind-data-migration-scripts/issues/330#issuecomment-2762205383

j-friedrich, 2025-04-01:
To match raw-qc, nwb and dff-qc now use the same naming convention: ROI0, ROI1, ROI2, ROI3 instead of Fiber1, etc. At least these three capsules (raw-qc, nwb-base, and fip-dff) are now consistent -- albeit, see Kenta's link above.

dougollerenshaw, 2025-04-01:
I'm still not following the details and implications of everything here. We're talking about the names of the extracted timeseries from the CMOS images from the two cameras (red and green), right? 

Overall, using zero-indexed names for everything makes sense. Kenta helped explain some technical details again yesterday. Just to be clear, we have the following, assuming that everything is zero-indexed:

* Four fibers, with names `fiber_0`, `fiber1`, `fiber_2`, `fiber_3`
* Five ROIs on each camera, each corresponding to a fiber, plus the background ROI positioned to avoid any of the fibers. These result in the following timeseries:
  * `G0`, `G1`, `G2`, `G3`, `G4` (all from the green camera when the blue LED is on)
  * `ISO0`, `ISO1`, `ISO2`, `ISO3`, `ISO4` (also from the green camera when the UV LED is on)
  * `R0`, `R1`, `R2`, `R3`, `R4` (all from the red camera when the yellow LED is on)

Is that correct?

And the issue here is that at some point, one-indexed names were used for at least some of these variables?

Questions:
* Where, if anywhere, are the one-indexed names still in use?
* Does this have implications only for the QC and processing pipelines, or does it also affect the variable names being written to the NWB files?
* If it does affect the variable names being written to the NWB files, does this mean we have an inconsistent naming convention across assets on S3?

@bwtan-allen said "I'm just wondering whether we have plans to unify these inconsistent naming in the past datasets. For now I think individual users can deal with this on their own end."

That implies that the data itself has this inconsistency. To me, having inconsistent naming conventions across assets on S3 sounds like a problem, if that's indeed what we're talking about here. Can someone help clarify?

dyf, 2025-04-01:
Let's use zero-based indexing everywhere consistently moving forward. 

I would prefer not to reprocess old data. Is there a clear signal in the metadata we can use to indicate the change? i.e. pipeline version?

j-friedrich, 2025-04-09:
> Is there a clear signal in the metadata we can use to indicate the change? i.e. pipeline version?

Yes, that changed with pipeline version 7. For better or worse, it’s one of the few versions where the CodeOcean version and the `processing.json` actually align — since the latter depends on manually updating the environment variable in the DFF capsule each time @Ahad-Allen makes a new release.

dougollerenshaw, 2025-05-09:
@dyf, I'm just circling back to this. Can you help me map out a path to closing this? 

@Ahad-Allen has a PR in for the pipeline readme. See here: https://github.com/AllenNeuralDynamics/aind-fiber-photometry-pipeline/pull/6

@j-friedrich answered your question about a clear signal for when data switched from 1-indexed to 0-indexed. Should I add a review comment to the PR asking this information be included in the readme? Just having it documented is a good first step, but it still makes analysis complicated because users will need an if-statement.

dougollerenshaw, 2025-05-09:
@bwtan-allen, is documenting this issue in the readme for the fiber pipeline a sufficient solution as far as you're concerned? Will it be reasonable to write analysis code with an if-statement on the pipeline version? If so, we should indicate that [Ahad's PR]([Readme AllenNeuralDynamics/aind-fiber-photometry-pipeline#6](https://github.com/AllenNeuralDynamics/aind-fiber-photometry-pipeline/pull/6)) will close this issue.

To me, it seems that having different indexing schemes in our released data is likely to lead to analysis mistakes. Documenting this in the pipeline is a minimal solution. Not sure how to weigh the cost tradeoff of analyst confusion vs the cost of reprocessing old data.

alexpiet, 2025-11-20:
Ideally, all NWB files would have 0 based indexing. We could live with older datasets using 1 based indexing as long as the metadata is consistent for each session (I believe it is, but I'm not confident). I personally think we would regret having the inconsistency - my belief is that the mental energy dealing with the inconsistency is much more expensive than the compute cost to reindex. 

I believe, but am not sure, that the raw data format never changed for DFT sessions. Rather the FIP preprocessing code changed. If that is true, then reprocessing the FIP data would correct the 1 based indexing to 0 based indexing. We also don't need to recompute the preprocessing steps. We could just write a script to rename things in the files. This shouldn't be a heavy lift. 

But I'm not the owner of the FIP platform, I'm just a user. I don't know who (if anyone) is in charge of photometry and can make this decision. @hagikent ?

I don't think updates to FIP hardware/acquisition would impact this decision, but tagging @bruno-f-cruz for visibility.

hagikent, 2025-11-21:
I'm on the same page as @alexpiet . Indeed, I was trying to make everything standardized with 0-index (vide supra).
We will be anyways reprocessing all photometry sessions for the alignment fix issue, which will solve this automatically?

alexpiet, 2026-04-10:
@rachelstephlee  @hagikent @bwtan-allen Has this been fixed with the reprocessing of all FIP data in Jan?

Do we know of any assets that are still using Fiber 1,2,3,4?

bwtan-allen, 2026-04-10:
me neither, thanks for checking.