AllenNeuralDynamics/aind-fip-dff #45: Figures are not visualized in QC

hagikent, 2025-04-02:
When names are changed (related to https://github.com/AllenNeuralDynamics/aind-fip-dff/issues/44), likely corresponding json was not updated correctly.

i.e. the following figures are not sent to QC but json still has ROI based namings for figures.
"dff-qc/Fiber0_poly.png" 
"dff-qc/Fiber1_poly.png" 
"dff-qc/Fiber2_poly.png"

hagikent, 2025-04-02:
below from @dbirman 
___
We have a little utility to help write migration scripts to fix records in docdb https://github.com/AllenNeuralDynamics/aind-data-migration-utils, I can help Johannes with it. If he knows the list of codeocean or docdb IDs and can write the migration function to replace the fields in the quality_control.json file

j-friedrich, 2025-04-02:
Fixed in PR [#46](https://github.com/AllenNeuralDynamics/aind-fip-dff/pull/46).
Will make it into the pipeline the next time @Ahad-Allen 
- merges PR
- syncs capsule
- releases capsule
- updates capsule version in pipeline layout to use newest one 
- releases pipeline

Ahad-Allen, 2025-04-17:
This is merged. Could you @j-friedrich  coordinate with dan on the script to update the old sessions that were affected? I can assist if your bandwidth is too limited.

j-friedrich, 2025-04-17:
```
from aind_data_migration_utils.migrate import Migrator
import argparse
import logging

logger = logging.getLogger("Migrate quality_control.json")
logger.setLevel(logging.INFO)

# Create a docdb query
query = {"name": "behavior_781897_2025-04-15_09-13-37_processed_2025-04-16_18-14-26"}

def your_callback(record: dict) -> dict:
    """
    Update metric names and references in a record's quality control section.

    This function replaces all occurrences of 'Fiber' with 'ROI' in the 'name' and
    'reference' fields of each metric in the Processing stage evaluations.

    Parameters
    ----------
    record : dict
        The document record containing quality control evaluations to be modified.

    Returns
    -------
    dict
        The modified record with updated metric names and references.
    """
    if "quality_control" not in record:
        logging.error("'quality_control' not in record")
    else:
        for evaluation in record["quality_control"]["evaluations"]:
            if evaluation["stage"] == "Processing":
                for metric in evaluation["metrics"]:
                    for key in ("name", "reference"):
                        metric[key] = metric[key].replace("Fiber", "ROI")
    return record

if __name__ == "__main__":
    parser = argparse.ArgumentParser()
    parser.add_argument("--full-run", action=argparse.BooleanOptionalAction, required=False, default=False)
    parser.add_argument("--test", action=argparse.BooleanOptionalAction, required=False, default=False)
    args = parser.parse_args()

    migrator = Migrator(
        query=query,
        migration_callback=your_callback,
        files=["quality_control"],
    )
    migrator.run(full_run=args.full_run, test_mode=args.test)
```

j-friedrich, 2025-04-17:
I don't know the list of codeocean or docdb IDs of the affected datasets

hagikent, 2025-04-17:
@j-friedrich 
>IDs of the affected datasets
I think they need to be queried based on **the version of the dff capsule**.
@Ahad-Allen what would be the best way to search assets based on the capsule version?

j-friedrich, 2025-04-18:
I think I figured out the correct docdb query we want. Edited above snippet. The dry run looks fine.

arielleleon, 2025-04-18:
Removing @Ahad-Allen since it sounds like @j-friedrich has this ready to go

dbirman, 2025-04-19:
I'll run the code from @j-friedrich in this PR https://github.com/AllenNeuralDynamics/aind-data-migration-scripts/pull/360 once I get a review from someone else familiar with the migration script pattern. This should be closed early next week.

hagikent, 2025-04-22:
Confirmed (not all of them but several)!

closed 2025-04-22 by hagikent

dbirman, 2025-04-22:
> Confirmed (not all of them but several)!

What does "not all of them" mean? Do we need to run this on additional sessions?

hagikent, 2025-04-22:
I just meant I haven't checked all of the sessions that got fixed.