AllenNeuralDynamics/aind-fip-dff #47: NWB files have both _dff-bright, and _proprocessed-bright

alexpiet, 2025-04-07:
This capsule preprocesses the fiber data. It appears to add both acquisition timeseries like `G_0_dff-bright` and `G_0_preprocessed-bright` .

Is this intentional? Are they meaningfully different? My understanding was that we were using `preprocessed` and depreciating `dff`

alexpiet, 2025-04-07:
Discussion with @hagikent:

dff: baseline correction
preprocessing: dff+ motion correction

alexpiet, 2025-04-07:
Lets make sure this NWB specification is what we want: https://alleninstitute-my.sharepoint.com/:w:/g/personal/alex_piet_alleninstitute_org/EchHIlpd_95DjrrDUuVTPuwBsPGGlfISoiMIieHlcWXbyw?e=2vQNWi

To close:
- [ ] Agreement on naming of `dff`, and `preprocessing` 
- [ ] Definitions of `dff` and `preprocessing` in the NWB specification file

j-friedrich, 2025-04-08:
I agree — having both in the NWB is confusing, and for downstream analysis, only the `preprocessed` trace should be used. If we're aligned on removing the `dff` trace from the NWB, I'm happy to make that change, while still keeping its panel in the QC plots, in order to help identify, in case of a failure, whether the issue occurred during the dF/F step or during motion correction.
The plot titles attempt to clarify the naming, but ideally the NWB itself should be self-explanatory.

hagikent, 2025-04-08:
From down-stream analysis point of view, I would keep both `preprocessed(w/o motion-correction)` and `preprocessed(w/motion-correction)`.
Some independent rationales:
1. when the motion signal is peculiar or massive motion-correction at times malfunctions 
2. when the isosbestic point of the sensor is off, motion-correction would under/over-estimate signals
3. etc.

I tag @ZhixiaoSu for her end-user opinion.

__
I was thinking of glossary and now view `preprocessed(w/o motion-correction)` and `preprocessed(w/motion-correction)` are least confusing, although lengthy. At least, `dff` is just one of multiple steps of the preprocessing and not suited to be used for representing the whole entity.

alexpiet, 2025-04-08:
Tagging @rachelstephlee and @bwtan-allen as end-users

rachelstephlee, 2025-04-08:
Yes johannes let me know the difference between the two as well. I think we just need documentation.

alexpiet, 2025-04-08:
I agree with @hagikent that including the processing without motion correction is valuable, and not particularly heavy to include.

I support the proposal for something like this, but I don't like "moco" because thats not intuitive
```
 <channel>_<fiber #>_preprocessed-<method>

 <channel>_<fiber #>_preprocessed-<method>-motion-corrected
``

dyf, 2025-04-08:
We should put English descriptions of these fields a) in the NWB packaging capsule in a configuration file, and b) in the NWB files themselves as properties of the TimeSeries object. 

https://pynwb.readthedocs.io/en/stable/pynwb.base.html#pynwb.base.TimeSeries

dyf, 2025-04-08:
We should also consider adding a table that associates field names with explicit properties so folks don't have to parse strings.

hagikent, 2025-04-08:
"moco", "MoCo", or "mc" seem to be common for folks including specialists in Flatiron (e.g. NoRMCorr), but agree, all jargon. no reservation against fully spelling-out.

dyf, 2025-04-08:
I think we can abandon the `preprocessed` monicker since the processed tracers are migrating to the `processing` module in NWB:

```
processing/<channel>_<fiber #>_<method>
processing/<channel>_<fiber #>_<method>_motion-corrected
``
@hagikent

j-friedrich, 2025-04-09:
We run 3 different dff-methods but only 1 mc-method for each, thus it could just be
```
processing/<channel>_<fiber #>_dff-<dff-method>
processing/<channel>_<fiber #>_dff-<dff-method>_motion-corrected
```
or
```
processing/dff-<dff-method>/<channel>_<fiber #>
processing/dff-<dff-method>/<channel>_<fiber #>_motion-corrected
```

More details, e.g. non-default parameter choices (albeit, at the moment they are not even exposed), would be found in the `processing.json`

hagikent, 2025-04-09:
I'm now confused on the scope of this discussion and the associating Doc file; which are we aiming?
1.only to consider current options
2.to be forgiving for highly possible future changes 

If 1, @j-friedrich 's suggestion would be perfectly fine.
If 2, given we are not satisfied with the current preprocessing and motion-correction, and new data will require further fine-tuning (as Sue/Bowen did), there is a fair chance of modifying regressor and regression methods, etc.

I naively thought 2 but please ignore if we only care 1.

dyf, 2025-04-10:
The only reason we would have to namespace/postfix the motion correction array name is if we want to store multiple versions of it, as we are doing with dF/F.

hagikent, 2025-04-10:
I foresee we will have different motioncorrection methods in the future and all should be stored in parallel.

j-friedrich, 2025-04-10:
Could also go in the other direction 😀 — if we ever settle on a single dF/F method.

One could rerun the pipeline with different methods as an _outer loop_ (as I do for ophys and slap2, mostly for development), resulting in one NWB per method. That seems more fitting if one method is the default and others are tested occasionally.

@alexpiet and @hagikent seem to prefer the _inner loop_ approach — running multiple methods in parallel within the pipeline. It has the benefit of flagging inconsistencies if preprocessing results differ significantly, or if downstream outcomes depend on the method used.

If we go that route and also want to support multiple motion correction methods, the structure would be as @dyf suggested:
```
processing/<channel>_<fiber #>_dff-<dff-method>
processing/<channel>_<fiber #>_dff-<dff-method>_motion-corrected-<mc-method>
```
For now, the only mc-method might be something like _iso-IRLS_ (isosbestic fit via IRLS).
As @hagikent noted, the regressor or regression method could also vary — e.g., switching from an isosbestic signal to a co-expressed sensor.

alexpiet, 2025-04-10:
I support the "inner loop" approach, and a naming scheme that supports the possibility of new motion correction methods
```
processing/<channel>_<fiber #>_dff-<dff-method>
processing/<channel>_<fiber #>_dff-<dff-method>_motion-corrected-<mc-method>
```

j-friedrich, 2025-04-10:
Ok, I tediously moved the preprocessing outputs from `acquistion` to `processing`. (NWB handles them surprisingly differently). For now the output looks like this:
[image]
Instead of adding directly to `processing` NWB insists on first adding a `ProcessingModule` to which then the traces can be added, hence the additonal layer in the directory hierachy called "fiber_photometry", which is not present for `acquistion`, i.e. `processing/fiber_photometry/<channel>_<fiber #>_dff-<dff-method>` vs `acquistion/<channel>_<fiber #>`.

One alternative would be to use the processing method as `ProcessingModule`, i.e.
`processing/dff-<dff-method>/<channel>_<fiber #>`
and
`processing/dff-<dff-method>_motion-corrected-<mc-method>/<channel>_<fiber #>`

If there is another `ProcessingModule` , e.g. "behavior", it probably makes sense to keep my current format.

Ahad-Allen, 2025-04-17:
Merged and deployed #48

closed 2025-04-17 by Ahad-Allen