AllenNeuralDynamics/aind-fip-dff #54: alignment off for fiber photometry validation has failures

rachelstephlee, 2025-08-25:
I found an alignment failure for the fiber photometry after the alignment fix. 

[image]

[image]

Found it for data asset [behavior_808057_2025-08-21_12-40-56_processed_2025-08-22_05-43-31](https://codeocean.allenneuraldynamics.org/data-assets/5c313ee6-c694-4796-9322-a94655511ad8/behavior_808057_2025-08-21_12-40-56_processed_2025-08-22_05-43-31?filters=N4IgjgrgpgTgniAXCArAYwMwEYNSgNgFo18BOAFkPIHZSjSMAmRwgQwvxS6y1YBMAHCAA0IAM4B7GABckINDCitpUPiJB8oYtEmkxoogA6sA5lCRYjsAAqnziRgAZR-AG6sAdmlVJgAX1EAWwgAG2kASwAxcLDYMV8AkA9WQPsQACMoAAtWV3CpAH0BR2KUagKnRhRCEsJGLAKsFnJHQhR8AsMYCW8xMVUKxyqagTrGAsdq8gxCbHVpU3jEAG0NWHDXH1Fi0up1TJy8qRAAXVEAa3CPNWRFMVDZUTFwgC97OkYMTgZRADMYrQWFCkFCiPjKVj9aRLZagcI3EC-FDpVgoJqtNCCUbTX58NhYbyEJrULD4VgCVhoRgEebhaQhNIHXL5GBFEqTcqVaq1eqNZqtdrqQISCAeWTIJlHVk7DmDYY8hpNKgC-DqTQMlQ3X6sEL9PxnECKLoSPgQNDhdIM3T6KBWGCBcJ9fIeJagPJQADu1oMIHdHoAyjlFN6oIkxAsVHJ-skQuoJB6PLAAJII37kfB8PhYPgzPjtVrkFB8X6EdLTUhsAQoKDkJRFvikdJxhOwAByKTSrCufAABP0YHlvOoFEpNQBBcVYahcATkLACRikAB0TFnC-wohR-QAqjBY8gstJpIYxIgAPRntAmqA9JQeJc6hkeRMQGA6vhwZIOtBiJdSExnuCCxQmIZ5phmWY5oQeb4AWRYlmWGAVuS1a1qimaNme6DYLgBDEGQlA0HQhAMMwbAcFwaK8IIw4SIEDrSCmchzukTSMNQ6R8Iu6S-J8XA0NQC4oBgvwYJSxbUI4hZlGg1Bkuo%2BgeP6CjhIY4ryNeZ6KcOrAnqEUBMcgjisFApCSSJJGOEq5B8I46RsIIeLpGSjglCgAjpMUaiiGgun3AyAAqcCGGkhiqVAIRXOYPl%2Bfp-ohBAJhyGUTRcBgOl6Qy7apHI0SZDAPbWFkEjSHRUB6HAhXdL0zweCYhXhZFiYZf5UBjmgNUAEIIIgzgaYEhgQBGzqGSApBImQmBoG03FUGm1BsOmMxQIwaC-ASFK4LZtEDUNyjOgA8i2MBRummbZrm%2BZUPBpblpWqF1hhTZGI1UXWNVWj9DCbpxM6FjUDFmUGQiNZlOQ1KULQThUKQValqwXx1PgjDpGgAikDg1DUKw6jGjV2VpL5QMFF21wFP8hgFB4HrpAUW5QAUROtY0jgtfpBNyKTeIU4Q1P2fTxCxVaAQ-TAzwSB4SACIDrWjWJplYFAxREnZRDpsChACFg5CsJrrC-HgDDAqwLS4x9fQc8gTP6ST3bk6pBRgGgBRvh6jRYGzWUdpz3aEDzTuEK7IAi76v0SxYMv6aNjgYOkYlrdzmglum6StPwysoI4pl8LWrkLnsRjm2IlvyELDNcwUZiJjA4TO5AOp0nAjMS3oEghAUUDuAl%2B0SyTJgmIoJjKIUOCe1AJdc4QVfrNN9eRdIcDEC33QhIQnc6hAPceGw-eD8PJ0h5sYt-YgGBGKwb6pCoYtILCSTexKlLnFA1zqF30ByC8F8H8IoBfmkQFWA9khFCHs-9g4GmtgyUanE0DjUcONGorkpi2WQvrCs1AlaMFYJxBg61SBm1vBbB%2BpdiZ806E-UwVwTAFHuOkAAVlANA0hGZl0aGPEurYADqnVCqUJMNQns-oIAMKYbIQ%2BYdJaIA3KQ2WCIMZYEcHmNA9lGCZkIpMJy%2BBTKEGoFUTMUkEHUCRIQ-GJCoHlzth%2BL8tdyZSCoXVR2zsmgcJIZPaxKRa5%2B3sQIuqhAnbB1-qHY%2B4dEBNEjtAhEsExL1HIBWZGWsqCsA4qWTiRBfhIkcL5LOjh8C1lMZ9EuFjbZkwpgUYsvwCgEIiePNxvseYVIgTUsQ25apJTCaIbsUAAAe-oIyEzooYDUWx5AQHDHRAAsuVfgEJfAaAhD2Bkmx9xrBrpsbyIAekhRrqkMUPYF4hTkJKFk6g6GMOYT2eEcgZRlECSAEIkJGLXB6T4MJ053JzjXEudMUk8kAxAJSCImwjrV2%2BiAK5yBwLnSgjBOCxZbpIXujWR6DZnr3xysgLmfZYCDmipswIXYVkvjfCEJcHjvxiAAAKPhflccMdIhpQD-DAJK%2BpOkeEGtIAAIrMlYcJUzIlROiYgWIqAiTxKwAkUAiTsVJOSSk1JVSiAiPSRk2RmSFBuZyIY3JRi8msiqIUIoxRHPVVKNkuw5S6rqIqfkbQlVrGGVqHUepIGKGUKoCcAAZHBEU5BjkSj2eowgg06qDYwRAmd1APPDEmZ53TVA%2BsyCsgN9Vg2hqqOGyNrMnivCgEmv1yAMZLixj2CZnUdoMWYlgVizAC7grEDy4C5UkDal1KGIAA)

j-friedrich, 2025-08-27:
> Please also see this session: b3359499-5d8e-47b2-9009-989843330087. GUI crashed closed to the end of the session, not sure this also resulted in misalignments.

Alignment looks correct here, but note it used the older pipeline (timestamps from `.json`).  
- `.json`: 86,868 timestamps
- `FIP....csv`: 88,650  
- `raw.harp`: 88,808  

So the last ~90s of fiber data were dropped. This wouldn’t have happened with the new pipeline.

j-friedrich, 2025-08-27:
> [@j-friedrich](https://github.com/j-friedrich) , would it be a fair plan that we will reprocess more DA sessions and give you more sessions that have this misalignment so that you can debug the source of this issue + how we can fix?

Instead of full reprocessing, I can just run a short script on all ~3000 raw data assets.  
(I already have one that produces the plots above, but I only ran it on the last 100.)  

There’s only real reason for concern if the number of timestamps from `raw.harp` and `FIP...csv` don’t agree.  
- In many cases, aligning the first FIP frame with the first harp timestamp works.  
- But there are exceptions:  
  - Extra harp timestamps **at the beginning** (e.g. `behavior_778234_2025-03-24_13-42-37`)  
  - Extra harp timestamps **at both beginning and end** (e.g. `behavior_778234_2025-03-28_13-43-40`)  

The key `fiber_photometry_start_time` usually lines up <1s before the first timestamp in `FIP...csv`.  
But in `behavior_778234_2025-03-24_13-42-37` it was **4 minutes prior**, which suggests missing FIP frames at the beginning — so aligning at the first harp timestamp won’t work.  

Next step: I’ll write a script that pulls out:  
- #, first, and last timestamp for `FIP...csv`, `raw.harp`, and `.json`  
- `fiber_photometry_start_time` and `fiber_photometry_end_time` from the `.json`  

(`fiber_photometry_start_time` / `end_time` are in daytime, as is `FIP...csv`.)

j-friedrich, 2025-09-05:
I created and ran the script I alluded to earlier, which generates a DataFrame with the following columns:
```
['fiber_photometry_start_time', 'fiber_photometry_end_time',
'Other_CurrentTime', 'Other_SessionStartTime', 'num raw', 'first raw',
'last raw', 'min_diff raw', 'max_diff raw', 'num json', 'first json',
'last json', 'min_diff json', 'max_diff json', 'num csv', 'first csv',
'last csv', 'min_diff csv', 'max_diff csv', 'num raw_cleaned',
'first raw_cleaned', 'last raw_cleaned', 'min_diff raw_cleaned',
'max_diff raw_cleaned', 'num csv_cleaned', 'first csv_cleaned',
'last csv_cleaned', 'min_diff csv_cleaned', 'max_diff csv_cleaned']
```
for all raw fiber sessions. This helps flag problematic sessions. The CSV can be found here: [timestamp_info2.csv](https://github.com/user-attachments/files/22163145/timestamp_info2.csv)

---

### Findings

- **Chattering not (anymore) an issue**  
  `min_diff raw` (minimal interval between raw timestamps) is never close to zero. The smallest observed (positive) interval across all datasets was **44.448 ms**, so the “chattering” concern mentioned [on slide 6](https://alleninstitute-my.sharepoint.com/:p:/g/personal/kenta_hagihara_alleninstitute_org/EWQRS4oGnDpJhIHheC8YJNABRSByVU_c3HRU1LdgwZ274g?e=6cZ7MX) does not appear to be a problem.

- **Misses very unlikely**  
  `max_diff raw` (maximal interval between raw timestamps) is almost never close to a multiple of 50 ms (what we’d expect if frames were missed). So misses are very unlikely.  
  When there *is* an unusually large gap in the raw timestamps, it is nearly always mirrored in the FIP CSV timestamps — consistent with the recording being stopped and restarted.  
  Our current cleanup, however, incorrectly assumes continuity: it fills the gap in the raw data and pretends no gap exists in the fiber data, which actively misaligns things.  
  Examples:  
  - `behavior_765228_2025-03-29_21-36-02` [image] [image]
  - `behavior_769036_2025-01-14_13-34-51` [image] [image]

- **Raw gaps not mirrored in CSV**  
  There are some cases where a large gap appears in the raw timestamps but not in the FIP CSV timestamps. This suggests the harp device was started, stopped, and restarted, with the FIP CSV only capturing the *later* portion.  
  - Sometimes this results in multiple FIP CSV files, e.g. `behavior_765622_2025-04-07_19-09-21`  [image] [image]
  - Sometimes not, e.g. `behavior_778234_2025-03-28_13-43-40` [image] [image]

In these cases, instead of filling the gap, we should remove the raw timestamps **prior to the gap**. This fixes alignment for both scenarios above.

For example, in `behavior_778234_2025-03-28_13-43-40`, neither “align at start” nor “align at end” worked (https://github.com/AllenNeuralDynamics/aind-fip-dff/issues/54#issuecomment-3221931021) — but dropping timestamps before the gap resolves the issue.
[image]

---

### In summary

- Look for unusually large gaps in the raw and FIP CSV timestamps.  
- Take the longest period with regularly spaced timestamps and truncate accordingly.  
- If the FIP timestamps were truncated, do the same with the FIP signal.
- Align at the start.